snu-csl/nvmevirt #6: Error inserting ZNS SSD kernel module

zy1024cs, 2023-04-06:
Hello, when I tried to enable CONFIG_NVMEVIRT_ZNS := y in the Makefile file and compiled it, the following error occurred when inserting the compiled file into the kernel module:
[image]
My /etc/default/grub file settings are as follows:
GRUB_CMDLINE_LINUX="memmap=8G\\\$2G intremap=off"
May I ask what may be the reason? thank you

arter97, 2023-04-06:
`sudo dmesg` (kernel log) will show more information.

Please post its output as well.

zy1024cs, 2023-04-07:
If memmap_size is set to a multiple of 1024, such as 8192M, an errors may occur in zns_ftl.c. The details are as follows:
[image]
Because **capacity** has been reduced by 1M in file main.c (The details are as follows), resulting in a value of 8191M.
[image]
When I try to set memmap_size to 8193M (8192+1) or 4097M (4096+1), this error will disappear, but another error will occur during insmod, like
[image]
Can you provide a suggested parameter that can make ZNS run? thank you

euidong-lee, 2023-04-12:
In your dmesg, It shows "ns_size=16384MiB".
It means that you set memmap_size to 16385M(16384 + 1).
Since your machine has only 10GB memory, this may cause issues. 
Can you check memmap_size?

euidong-lee, 2023-04-12:
Thanks to your help, we find memory corruption when accessing the struct zone_report.
Could you modify the code as follows and try it?
[image]

We just use 'zonemode=zbd' option for fio. 
The other options are same for testing conventional ssds.
Please note that the write unit size for ZNS_PROTOTYPE is set to 128KB, due to the constraints of the prototype SSD that we used. So, the write request should be a multiple of 128KB. Also, write buffering is not supported in this prototype SSD, which means that write latency exposes the NAND program time.

arter97, 2023-04-13:
@zy1024cs, please try out #7.

arter97, 2023-04-13:
Linux kernel doesn't support ZNS without Zone Append capability.

Read this thread for context:
https://lore.kernel.org/all/20200818052936.10995-1-joshi.k@samsung.com/T/#u

If you don't use ZA, you can probably just fake NVMeVirt to report it's supported to make it R/W.
Try uncommenting this line: https://github.com/snu-csl/nvmevirt/pull/7/commits/96f21b9936f88800c2ab85e6eec2f824ccd5779a#diff-c714e0d9e04cd3a91b0e2de1c353296a3550544d3c5c63ff18f129f974beaa25R224

Ideally, it'd be nice if NVMeVirt adds support for ZA :upside_down_face:

PR #7 (2023-04-19): ZNS initialization fixes
The existing get_log_page implementation can trigger malfunctions, especially under the ZNS configuration.

This PR addresses it by fixing memory accesses and implementing NVME_LOG_CMD_EFFECTS required by Linux's NVMe device driver.

Also, off-by-one memory allocation issue has been addressed.

I believe this will fix #6.

Thanks.
closed 2023-04-19 by beowulf