snu-csl/nvmevirt #9: License

jinsoox, 2023-04-21:
The license has been changed to GPL v3. We need to change the License section at the end of the README.md file to reflect this change (both the license version and the associated link).

arter97, 2023-04-21:
GPL v3 doesn't work nicely with the Linux kernel.

All GPL exported functions within the Linux kernel cannot be used when the module itself doesn't allow GPL v2. This includes GPL v3. In other words, GPL 3.0-only is treated the same as a proprietary license in perspective of the Linux kernel and modules.

Also, in case the authors are planning to upstream this to the mainline Linux kernel in the long-run, the use of GPL v3 will immediately make it un-upstreamable and re-licensing it back to GPL v2 will be a hassle as more people contribute code to this repository. There isn't a single upstreamed Linux kernel driver that uses GPL v3.

I hope the authors understand the ramification of this change. I strongly recommend sticking with GPL-2.0. If you really want downstream to use this code as GPL v3, please consider GPL-2.0+.

beowulf, 2023-04-24:
Agreed, I switched the license too carelessly. NVMeVirt will go GPL-2.

@euidong-lee Would you like to check the license of pqueue.[ch], which is not under the GPL license.

arter97, 2023-04-24:
> Agreed, I switched the license too carelessly. NVMeVirt will go GPL-2.
> 

Good to know, I'd recommend force-pushing HEAD^ instead of reverting the HEAD commit to avoid further confusion.

My recent PRs are made from HEAD^.

> @euidong-lee Would you like to check the license of pqueue.[ch], which is not under the GPL license.

BSD 2-clause is compatible with GPL: https://www.gnu.org/licenses/license-list.html#FreeBSD
Nothing to worry here except for "2. Redistributions in binary form must reproduce the above copyright notice". I'll send another PR for this.

beowulf, 2023-04-24:
Thanks a lot @arter97.