snu-csl/nvmevirt #14: SPDK support

Krien, 2023-05-09:
Hello, does NVMeVirt currently have SPDK support?
NVMeVirt works fine on my machines for the NVM, ZNS, and SSD configs when used in the kernel. However, whenever I unbind the virtual device from the kernel, I can not get it to bind to another driver (`vfio`,  `uio_pci_generic`) or even the original `nvme` driver.  It fails with the following messages (debugging enabled):
[image]
As a result the device can no longer be used.
To reproduce (change with relevant address):
```bash
echo 0001:10:00.0 | sudo tee /sys/bus/pci/devices/0001\:10\:00.0/driver/unbind
echo 0001:10:00.0 | sudo tee /sys/bus/pci/drivers/nvme/bind
```
or:
```bash
sudo ./spdk/scripts/setup.sh
sudo ./spdk/scripts/setup.sh reset
```
We have disabled intremap and make use of Linux kernel `5.19.2`. Note that currently it is run from within QEMU (6.1.0) since we can not deploy it on the host machine. 

Is this unbinding/binding a known issue? If so, is there a way to solve it or an alternative to use SPDK?
Thank you in advance!

arter97, 2023-05-09:
NVMeVirt currently doesn't support re-enabling an NVMe instance, which is what you'd be hitting if you're rebinding it to SPDK.

Try blacklisting NVMe kernel module itself (adding `lsmod | grep nvme` to /etc/modprobe.d/blacklist.conf) so that the VFIO or UIO would be the first to use the device.

jaehshim, 2023-05-09:
NVMeVirt currently does have SPDK support and it works on both physical machines and FEMU.
Therefore, the below script should work fine.

```bash
sudo ./spdk/scripts/setup.sh
sudo ./spdk/scripts/setup.sh reset
```

@arter97 did you face the same issue as @Krien ?

Krien, 2023-05-09:
Thank you both for the explanation! Unfortunately this leads to another error.
If I do (after blacklisting):
```bash
 sudo PCI_ALLOWED="none"  ./spdk/scripts/setup.sh # hugepage only
 sudo modprobe uio_pci_generic
 echo "0c51 0101" | sudo tee /sys/bus/pci/drivers/uio_pci_generic/new_id # Ensure uio will be used later
 sudo insmod ./nvmev.ko memmap_start=32G memmap_size=32G cpus=3,4
```
And then try to use SPDK, e.g. the hello world:
```
sudo ./spdk/build/examples/hello_world
```
It leads to a kernel panic.
[image]
I get the same issue with `vfio-pci`.
 
What SPDK version, FEMU and kernel version did you test with @jaehshim? It might be local to my configuration.

Krien, 2023-05-10:
I have indeed added a few `NVMEV_INFO` lines, but  (I hope) those should not interfere too much.  After resetting the repository the issue is still there. I have tested it on a newer kernel (6.1.2), different VM ans machine as well , but this did not  seem to resolve the error.  I think you are right that it is an initialization error.
The full nvmev code called is:
```
[   60.507732] NVMeVirt: Storage : 800100000 + 780000000
[   60.558431] NVMeVirt: [NVMEV_NAMESPACE_INIT] ns=0 ns_addr=000000009550afe1 ns_size=30720(MiB)
[   60.558535] PCI host bridge to bus 0001:10
[   60.558555] pci_bus 0001:10: root bus resource [io  0x0000-0xffff]
[   60.558559] pci_bus 0001:10: root bus resource [mem 0x00000000-0x3fffffffffff]
[   60.558561] pci_bus 0001:10: root bus resource [bus 00-ff]
[   60.558574] pci 0001:10:00.0: [0c51:0101] type 00 class 0x010802
[   60.558579] pci 0001:10:00.0: reg 0x10: [mem 0x800000000-0x800003fff 64bit]
[   60.558582] pci 0001:10:00.0: reg 0x30: [mem 0x00000000-0x000007ff pref]
[   60.558585] pci 0001:10:00.0: enabling Extended Tags
[   60.569541] NVMeVirt: Successfully created virtual PCI bus (node 0)
[   60.570472] NVMeVirt: nvmev_io_worker_0 started on cpu 4 (node 0)
[   60.570612] NVMeVirt: Successfully created Virtual NVMe deivce
[   60.570652] NVMeVirt: nvmev_dispatcher started on cpu 3 (node 0)
[   82.988487] uio_pci_generic 0001:10:00.0: No IRQ assigned to device: no support for interrupts?
[   83.052586] uio_pci_generic 0001:10:00.0: BAR 6: assigned [mem 0xfec00800-0xfec00fff pref]
[  137.367242] BUG: kernel NULL pointer dereference, address: 0000000000000038
[  137.367298] #PF: supervisor read access in kernel mode
[  137.367323] #PF: error_code(0x0000) - not-present page
```
The full call trace that follows is:
```
[  137.367843] Call Trace:
[  137.367865]  <TASK>
[  137.367899]  xas_load+0x1f/0xf0
[  137.367925]  ? kvm_sched_clock_read+0x2/0x40
[  137.367953]  xas_find+0x184/0x1e0
[  137.367986]  xa_find+0xc6/0xe0
[  137.368028]  nvmev_signal_irq+0x54/0xd0 [nvmev]
[  137.368055]  nvmev_proc_admin_sq+0x845/0xc50 [nvmev]
[  137.368080]  ? __proc_file_read+0x230/0x230 [nvmev]
[  137.368103]  nvmev_dispatcher+0x90/0x250 [nvmev]
[  137.368127]  kthread+0xeb/0x120
[  137.368158]  ? kthread_complete_and_exit+0x20/0x20
[  137.368192]  ret_from_fork+0x1f/0x30
[  137.368226]  </TASK>
```
Did you happen to use `uio_pci_generic` as well? If so, what steps are missing from the log?

jaehshim, 2023-05-22:
Sorry, I seems that I have misunderstood your log and thought it was an initialization error.
It actually looks like there is something wrong while device identifying.

Could you please check which line the NULL pointer dereference happens? 
My guess is line 152 in function `__nvmev_admin_identify_ctrl()`.

Krien, 2023-05-22:
No problem. It fails on `nvmev_signal_irq(0)`,  line 574 in `admin.c`.  The page fault is on the msi xarray.
It does not seem to have problems with `__nvmev_proc_admin_req` itself.
It successfully sends an identify command for the controller, but it can not send the interrupt.

beowulf, 2023-05-23:
Seems like the issue is for handling irq through msi-x. Could you check whether the VM/guest kernel set to support MSI-X? 
Also could you try to use CONFIG_NVMEV_FAST_X86_IRQ_HANDLING in nvmev.h? It is undefined by default, so change it to #define. 

#undef CONFIG_NVMEV_FAST_X86_IRQ_HANDLING

Krien, 2023-05-23:
Thank you for the continued support @beowulf  and @jaehshim .

The QEMU VM itself supports MSI-X. NVMeVirt can be used correctly with the `NVMe` module.  Both `lspci -vv`  and  `cat /proc/interrupts` show that MSI-X is enabled in this situation. 

The issue only arises when `uio_generic_pci` or `vfio-pci` (requires `enable_unsafe_noiommu_mode` since iommu is not supported)  are used. These are the drivers that are typically used for SPDK.  For both drivers there is no support for MSI-X according to `lspci -vv`.  This is the likely issue.
It might be that these drivers do not allow MSI-X by default without IOMMU? Am I missing a step/configuration here that is needed for NVMeVirt? Some flags in `/boot/config/...` or in `/sys/module/vfio/parameters/`? 

With the define the same issue arises. Probably because the msi-x vector is not allocated.  It fails on line 65 of `pci.c`,  
```C
xa_for_each_range(xa, idx, msi_desc, 0, NR_MAX_IO_QUEUE * PCI_MSIX_ENTRY_SIZE) {
```

beowulf, 2023-07-07:
Turns out uio_pci_generic requires INTx handling, which was excluded while code cleanup. Would you like to test with e9bbeaf which reverts the removal?

Krien, 2023-07-07:
Thank you for the update @beowulf .
I have rerun the code (with the new fix), exact same setup and I get a different error now. Fortunately APIC now does work :). The erro is with both with `uio_generic_pci` and `vfio`.  Unfortunately can not debug this more in-depth at the moment. The error is:
```
[   63.453677] kernel BUG at /home/user/src/nvmevirt/pci.c:92!
```
This has to do with `pdev->msix_enabled` being false and `vdev->intx_disabled` being true.
I am not sure how this can happen, I assume that some flag is set incorrectly somewhere leading to neither MSIX or INTX being enabled. For reference, I have logged all masks before the bug occurred:
```
[   40.515591] NVMeVirt: [NVMEV_NAMESPACE_INIT] ns=0 ns_addr=000000008e9c1964 ns_size=30623(MiB)
[   40.515695] PCI host bridge to bus 0001:10
[   40.515701] pci_bus 0001:10: root bus resource [io  0x0000-0xffff]
[   40.515704] pci_bus 0001:10: root bus resource [mem 0x00000000-0x3fffffffffff]
[   40.515706] pci_bus 0001:10: root bus resource [bus 00-ff]
[   40.515720] pci 0001:10:00.0: [0c51:0101] type 00 class 0x010802
[   40.515722] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x402, size: 2, new: 0x402
[   40.515724] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x402 -> 0x2, size: 2, new: 0x2
[   40.515727] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515729] NVMeVirt: [W] 0x10, mask: 0xffffc000, val: 0x4 -> 0xffffffff, size: 4, new: 0xffffc004
[   40.515731] NVMeVirt: [W] 0x10, mask: 0xffffc000, val: 0xffffc004 -> 0x4, size: 4, new: 0x4
[   40.515732] NVMeVirt: [W] 0x14, mask: 0x0, val: 0x8 -> 0xffffffff, size: 4, new: 0x8
[   40.515734] NVMeVirt: [W] 0x14, mask: 0x0, val: 0x8 -> 0x8, size: 4, new: 0x8
[   40.515735] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515737] pci 0001:10:00.0: reg 0x10: [mem 0x800000000-0x800003fff 64bit]
[   40.515739] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515740] NVMeVirt: [W] 0x18, mask: 0x0, val: 0x0 -> 0xffffffff, size: 4, new: 0x0
[   40.515741] NVMeVirt: [W] 0x18, mask: 0x0, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515742] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515743] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515744] NVMeVirt: [W] 0x1c, mask: 0x0, val: 0x0 -> 0xffffffff, size: 4, new: 0x0
[   40.515745] NVMeVirt: [W] 0x1c, mask: 0x0, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515746] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515747] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515748] NVMeVirt: [W] 0x20, mask: 0x0, val: 0x0 -> 0xffffffff, size: 4, new: 0x0
[   40.515749] NVMeVirt: [W] 0x20, mask: 0x0, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515750] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515751] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515752] NVMeVirt: [W] 0x24, mask: 0x0, val: 0x0 -> 0xffffffff, size: 4, new: 0x0
[   40.515753] NVMeVirt: [W] 0x24, mask: 0x0, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515754] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515755] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x0, size: 2, new: 0x2
[   40.515756] NVMeVirt: [W] 0x30, mask: 0x0, val: 0x0 -> 0xfffff800, size: 4, new: 0x0
[   40.515757] NVMeVirt: [W] 0x30, mask: 0x0, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515759] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x2, size: 2, new: 0x2
[   40.515761] pci 0001:10:00.0: enabling Extended Tags
[   40.515762] NVMeVirt: [W] 0x68, mask: 0xffffffff, val: 0x0 -> 0x100, size: 2, new: 0x100
[   40.515786] NVMeVirt: [W] 0x52, mask: 0xc000, val: 0x807f -> 0x7f, size: 2, new: 0x7f
[   40.515802] NVMeVirt: [W] 0x110, mask: 0xffffffff, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.515803] NVMeVirt: [W] 0x104, mask: 0xffffffff, val: 0x0 -> 0x0, size: 4, new: 0x0
[   40.522150] NVMeVirt: Successfully created virtual PCI bus (node 0)
[   40.523433] NVMeVirt: nvmev_io_worker_0 started on cpu 4 (node 0)
[   40.523662] NVMeVirt: Successfully created Virtual NVMe device
[   40.523752] NVMeVirt: nvmev_dispatcher started on cpu 3 (node 0)
[   62.849202] ACPI: \_SB_.LNKA: Enabled at IRQ 11
[   63.452634] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x6, size: 2, new: 0x2
[   63.453123] NVMeVirt: [W] 0x4, mask: 0x400, val: 0x2 -> 0x406, size: 2, new: 0x402
[   63.453675] ------------[ cut here ]------------
```

beowulf, 2023-07-10:
@Krien, Would you like to test with the current main?
To tell the full story, in the past, we used the software interrupt to handle INTx and such way was deprecated while we cleaning up the code for publication. So we simply removed them. In the last patch, I tried to revert them by grabbing some part of previous implementation, which does not work properly. So I reworked the INTx handling to the modernized irq handling, and confirmed that worked in my testing bed.

beowulf, 2023-07-10:
@Krien, nevermind, the latest patch still does not make it work on a real device (works well on VMs though).

beowulf, 2023-07-11:
Turned out that the SPDK issue was originated from multiple changes that are made while cleaning up the code.

1. INTx handling
As mentioned above, intx handling was removed by mistake. It is reverted with a better/nicer/cleaner abstraction.

2. AQA resetting
While cleaning up the BAR processing code for allowing driver unloading,  AQA handling was not properly refactored so ASQ/ACQ head is not properly reset on AQA change. This caused when the SPDK inserted an requests in ASQ after resetting the AQA, the dispatcher cannot catch the request, making it halt. This is changed

4. BAR change handing
The host device driver sets up the device by setting the NVMe control registers. Each change is delivered to the device through a PCI transaction, allowing the device to process the requests in the requested order. As the change can be detected by the looping dispatcher, multiple changes can be 'queued' without preserving the order between the changes. Now the dispatcher handles it with an 'expected' order.

closed 2023-07-20 by beowulf

Krien, 2023-08-08:
@beowulf, I appreciate the code fixes and the attached explanation. 
 I see that the issue is already closed, but for the sake of completeness I confirmed that it works as well on my VMs (both in QEMU and FEMU). Thank you for the continued development!

beowulf, 2023-08-10:
@Krien Good to hear that it works fine on your environment. Thanks for your reporting and testing!

commit e9bbeaf8 (2023-07-07): pci: Support the INTx mechanism

INTx was removed while cleaning up code for publication, but turns out
it is required for uio_pci_generic. This patch puts it back hoping to
resolve #14 and #17.
commit 66bf1d3e (2023-07-10): pci: Support the INTx mechanism

INTx was removed while cleaning up code for publication, but turns out
it is required for uio_pci_generic. This patch puts it back hoping to
resolve #14 and #17.
commit 234bbf5a (2023-07-11): pci: Support the INTx mechanism

INTx was removed while cleaning up code for publication, but turns out
it is required for uio_pci_generic. This patch puts it back hoping to
resolve #14 and #17.