snu-csl/nvmevirt #16: Is CMB supported in nvmevirt for P2PDMA?

ingerido, 2023-06-08:
Hi,

    This is really an amazing work! I'm wondering if CMB (controller memory buffer) supported in nvmevirt to perform peer-to-peer data transfer? I did not see this part in README, and it seems not supported from what lspci tells:
    
```
	Latency: 0
	NUMA node: 0
	Region 0: Memory at 800000000 (64-bit, non-prefetchable) [size=16K]
	Expansion ROM at <unassigned> [disabled] [size=2K]

```

    If CMB not supported, how should I run peer-to-peer data transfer between virtual nvme devices?

Thanks!

jaehshim, 2023-06-08:
Hello,

Currently CMB is not supported and we are implementing the functionality to support them.

Without CMB, the P2P data transfer works like any other NVMe SSDs that doesn't have CMB support.
For example, NVIDIA GDS will make the SSD to transfer data to the GPU memory, not the host memory.

ingerido, 2023-06-08:
Thanks for the reply. I mean even for P2PDMA framework in Linux kernel, if I want to do data transfer between 2 nvme storage devices, at lease 1 nvme storage need to have a CMB registered and published as a memory for performing P2PDMA.

jaehshim, 2023-06-09:
Ah in that case BAR memory can be used instead of CMB. (GDS is using this method)

"This is currently the only arrangement supported by the kernel but one could imagine slight tweaks to this that would allow for the same functionality. For example, if a specific RNIC added a BAR with some memory behind it, its driver could add support as a P2P provider and then the NVMe Target could use the RNIC's memory instead of the CMB in cases where the NVMe cards in use do not have CMB support." from https://docs.kernel.org/driver-api/pci/p2pdma.html

ingerido, 2023-06-09:
Got it. Thanks for the reply!

closed 2023-06-09 by ingerido