snu-csl/nvmevirt #25: Listing nvmevirt device using xnvme

MarcusParadies, 2024-01-11:
Hi!

I've been playing around with nvmevirt and wanted to create a small demo app using [xnvme](https://xnvme.io/) and nvmevirt. I can create a virtual device and see it through `lsblk` and `sudo nvme list`.

When I try to issue `sudo xnvme enum`, the execution blocks for a few seconds, then lists the device. When I then run `sudo nvme list` again, the output look like this: 

[image]

The virtual device is still somehow there but not functional anymore. I was wondering whether anyone has experienced something like this before. Was also wondering if this issue is related to nvmevirt or xnvme.

Thanks!

beowulf, 2024-01-31:
NVMeVirt only expects that the host queries device identity with 0x05 and 0x06 CNS value for ZNS SSD not for general SSD. However xnvme tool utilizes the feature, making the exception

MarcusParadies, 2024-01-31:
Thanks a lot!