snu-csl/nvmevirt #30: nvmev_io_worker error

mtm776, 2024-04-01:
Using fio(with job file - # of session 8 / depth = 32 / bs = 256K / time base), this module generates kernel panic with below kernel message

[ 4892.002303] BUG: scheduling while atomic: nvmev_io_worker/21188/0x00000002
[ 4892.002409] CPU: 65 PID: 21188 Comm: nvmev_io_worker Tainted: G           OE      6.2.5 #2
[ 4892.002413] Call Trace:
[ 4892.002415]  <TASK>
[ 4892.002419]  dump_stack_lvl+0x48/0x70
[ 4892.002430]  dump_stack+0x10/0x20
[ 4892.002432]  __schedule_bug+0x64/0x80
[ 4892.002439]  __schedule+0xef7/0x14c0
[ 4892.002446]  ? asm_sysvec_apic_timer_interrupt+0x1b/0x20
[ 4892.002452]  ? asm_sysvec_apic_timer_interrupt+0x1b/0x20
[ 4892.002454]  schedule+0x68/0x110
[ 4892.002456]  schedule_preempt_disabled+0x15/0x30
[ 4892.002459]  __mutex_lock.constprop.0+0x6f4/0x770
[ 4892.002461]  ? sysvec_apic_timer_interrupt+0x4e/0xb0
[ 4892.002464]  ? asm_sysvec_apic_timer_interrupt+0x1b/0x20
[ 4892.002466]  __mutex_lock_slowpath+0x13/0x20
[ 4892.002467]  mutex_lock+0x3e/0x50
[ 4892.002469]  msi_domain_get_virq+0xf3/0x160
[ 4892.002475]  nvmev_signal_irq+0x2d/0xa0 [nvmev]
[ 4892.002483]  nvmev_io_worker+0x900/0xb10 [nvmev]
[ 4892.002489]  ? __pfx_nvmev_io_worker+0x10/0x10 [nvmev]
[ 4892.002494]  kthread+0xeb/0x120
[ 4892.002498]  ? __pfx_kthread+0x10/0x10
[ 4892.002501]  ret_from_fork+0x29/0x50
[ 4892.002506]  </TASK>

why this error occur during fio? Is some method to solve this problem?

arter97, 2024-04-01:
Can you try changing
```
#undef CONFIG_NVMEV_FAST_X86_IRQ_HANDLING
```
to
```
#define CONFIG_NVMEV_FAST_X86_IRQ_HANDLING
```
in nvmev.h?

mtm776, 2024-04-01:
I change this part in nvmev.h, but this error still occurs.

arter97, 2024-04-01:
@mtm776 Can you try #33?

mtm776, 2024-04-01:
After changing the code, fio runs well without error. Thank you!

closed 2024-04-01 by mtm776

arter97, 2024-04-01:
Great, thanks for the confirmation!

PR #33 (2024-05-30): Convert irq_lock to mutex
MSI helper functions within the Linux kernel may try to catch mutex [1][2] while we hold a spinlock with preemption disabled.

Fixes: #30
Link: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/kernel/irq/msi.c?h=v6.6#n448 [1]
Link: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/kernel/irq/msi.c?h=v6.6#n334 [2]
commit 9bd1300f (2024-05-30): Convert irq_lock to mutex

MSI helper functions within the Linux kernel may try to catch mutex [1][2]
while we hold a spinlock with preemption disabled.

Fixes: #30
Link: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/kernel/irq/msi.c?h=v6.6#n448 [1]
Link: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/kernel/irq/msi.c?h=v6.6#n334 [2]