snu-csl/nvmevirt #34: mount: /home/virgo/nvmevirt/test: wrong fs type, bad option, bad superblock on /dev/nvme1n1, missing codepage or helper program, or other error.

Evanshenf, 2024-04-07:
JBD2: no valid journal superblock found
[  449.510702] EXT4-fs (nvme1n1): error loading journal

I am using Ubuntu 20.04 under VMware, with kernel version 5.15-0-58. After compiling with make, insmod has no problems. However, after executing mkfs.ext4 /dev/nvme1n1, I encounter this error when attempting to mount.

Then I uninstalled nvme again and ran modprobe nvme, and the following error occurred.

[  726.076392] nvme nvme1: I/O 0 QID 0 timeout, disable controller
[  726.276816] nvme nvme1: Identify Controller failed (-4)
[  726.276822] nvme nvme1: Removing after probe failure status: -5

Evanshenf, 2024-04-07:
But the write and read operations work normally with my custom-written nvme driver.
[image]

jaehshim, 2024-04-11:
In normal cases, mkfs.ext4 should work fine.

Could you add some more details about how you have encountered this error?
Have you tried mkfs.ext4 right after insmod?

closed 2024-05-23 by jaehshim