snu-csl/nvmevirt #35: Does NVMeVirt support multiple SSDs？

THU-jty, 2024-04-13:
I want to emulate multiple SSDs on my server and compile multiple kernel modules. However, it reports that "Unable to create PCI bus". 

I wonder if NVMeVirt supports emulating multiple SSDs now. In FAST'23 paper, it seems that the feature is not ready yet. If it is enabled now, how to emulate multiple SSDs? Thanks.

jaehshim, 2024-04-15:
Currently supporting multiple SSDs are on-development.

However, you can emulate multiple SSDs by loading multiple NVMeVirt modules.
As long as you make the module name and PCIe bus number(NVMEV_PCI_BUS_NUM) unique, it would work.

Please let me know if you face any other challenges while loading multiple NVMeVirt modules :)

jaehshim, 2024-04-15:
https://github.com/jaehshim/nvmevirt/tree/multi
This is a simple example code of supporting multiple (max = 4) NVMeVirt modules.
Check `init_nvmev.sh` to see how each module is built and loaded
(Keep in mind that you should change the insmod parameters)

THU-jty, 2024-04-25:
Thanks very much for your detailed feedback!  It works and now I run multiple NVMeVirt modules successfully.

closed 2024-04-25 by jaehshim