snu-csl/nvmevirt #41: ssd: Array index is out of range with CELL_MODE_TLC

abcinje, 2024-05-27:
In the following function, the return value can vary from 0 to 3
if the `CELL_MODE` has been defined as `CELL_MODE_TLC` in `ssd_config.h`.
```
static inline uint32_t get_cell(struct ssd *ssd, struct ppa *ppa)
{
	struct ssdparams *spp = &ssd->sp;
	return (ppa->g.pg / spp->pgs_per_flashpg) % (spp->cell_mode + 1);
}
```

In the code below, `pg_4kb_rd_lat` and `pg_rd_lat` are arrays with length 3 (which is from `MAX_CELL_TYPES`)
So when the return value of `get_cell()` is 3, it invokes out of range error.
```
uint64_t ssd_advance_nand(struct ssd *ssd, struct nand_cmd *ncmd)
{
	...
	spp = &ssd->sp;
	lun = get_lun(ssd, ppa);
	ch = get_ch(ssd, ppa);
	cell = get_cell(ssd, ppa);
	remaining = ncmd->xfer_size;

	switch (c) {
	case NAND_READ:
		/* read: perform NAND cmd first */
		nand_stime = max(lun->next_lun_avail_time, cmd_stime);

		if (ncmd->xfer_size == 4096) {
			nand_etime = nand_stime + spp->pg_4kb_rd_lat[cell];
		} else {
			nand_etime = nand_stime + spp->pg_rd_lat[cell];
		}
	...
}
```

I think the right way would be to divide by `spp->cell_mode` instead of `spp->cell_mode + 1`.

beowulf, 2024-05-30:
It seems there was a mistake while switching to public repository. Thanks.

jaehshim, 2024-05-30:
Fixed in commit 07acdd2.

closed 2024-05-30 by jaehshim