snu-csl/nvmevirt #43: A compilation error after setting `NR_NAMESPACES` to 2

yj-han, 2024-06-03:
I attempted to compile NVMeVirt with `BASE_SSD` set to `ZNS_PROTOTYPE`. I successfully compiled the program and initialized NVMe when I set `NR_NAMESPACES` to 1. However, I encountered compilation errors when `NR_NAMESPACES` was set to 2. 

To configure the setup, I modified the `Kbuild` file by setting `CONFIG_ZNS := y` and uncommenting `ccflags-$(CONFIG_NVMEVIRT_ZNS) += -DBASE_SSD=ZNS_PROTOTYPE`. Additionally, I updated the `ssd_config.h` file to reflect the desired value for `NR_NAMESPACES`.

Are there any configurations that I may have overlooked?

The following lines contain the error messages I got:

```sh
make -C /lib/modules/5.15.0-107-generic/build M=/home/os/nvmevirt modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-107-generic'
  CC [M]  /home/os/nvmevirt/main.o
  CC [M]  /home/os/nvmevirt/pci.o
  CC [M]  /home/os/nvmevirt/admin.o
  CC [M]  /home/os/nvmevirt/io.o
  CC [M]  /home/os/nvmevirt/dma.o
  CC [M]  /home/os/nvmevirt/ssd.o
  CC [M]  /home/os/nvmevirt/zns_ftl.o
  CC [M]  /home/os/nvmevirt/zns_read_write.o
  CC [M]  /home/os/nvmevirt/zns_mgmt_send.o
  CC [M]  /home/os/nvmevirt/zns_mgmt_recv.o
  CC [M]  /home/os/nvmevirt/channel_model.o
  LD [M]  /home/os/nvmevirt/nvmev.o
  MODPOST /home/os/nvmevirt/Module.symvers
ERROR: modpost: "kv_init_namespace" [/home/os/nvmevirt/nvmev.ko] undefined!
ERROR: modpost: "simple_init_namespace" [/home/os/nvmevirt/nvmev.ko] undefined!
ERROR: modpost: "conv_init_namespace" [/home/os/nvmevirt/nvmev.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:133: /home/os/nvmevirt/Module.symvers] Error 1
make[2]: *** Deleting file '/home/os/nvmevirt/Module.symvers'
make[1]: *** [Makefile:1830: modules] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-107-generic'
make: *** [Makefile:8: default] Error 2
```

jaehshim, 2024-06-13:
It seems that the code needs some modifications to fix the compile error.

In the mean time, if you are not planning to use kv_ftl, you can change the Kbuild file as below and remove the code that uses kv_ftl ([main.c:L541](https://github.com/snu-csl/nvmevirt/blob/07acdd24e03d699bc88b272966d083bad21c82c6/main.c#L541), [main.c:L569](https://github.com/snu-csl/nvmevirt/blob/07acdd24e03d699bc88b272966d083bad21c82c6/main.c#L569)) to avoid the compile error.

```
# Select one of the targets to build
#CONFIG_NVMEVIRT_NVM := y
#CONFIG_NVMEVIRT_SSD := y
CONFIG_NVMEVIRT_ZNS := y
#CONFIG_NVMEVIRT_KV := y

obj-m   := nvmev.o
nvmev-objs := main.o pci.o admin.o io.o dma.o
ccflags-y += -Wno-unused-variable -Wno-unused-function

ccflags-$(CONFIG_NVMEVIRT_NVM) += -DBASE_SSD=INTEL_OPTANE
nvmev-$(CONFIG_NVMEVIRT_NVM) += simple_ftl.o

ccflags-$(CONFIG_NVMEVIRT_SSD) += -DBASE_SSD=SAMSUNG_970PRO
nvmev-$(CONFIG_NVMEVIRT_SSD) += ssd.o conv_ftl.o pqueue/pqueue.o channel_model.o

ccflags-$(CONFIG_NVMEVIRT_ZNS) += -DBASE_SSD=WD_ZN540
#ccflags-$(CONFIG_NVMEVIRT_ZNS) += -DBASE_SSD=ZNS_PROTOTYPE
ccflags-$(CONFIG_NVMEVIRT_ZNS) += -Wno-implicit-fallthrough
nvmev-$(CONFIG_NVMEVIRT_ZNS) += ssd.o zns_ftl.o zns_read_write.o zns_mgmt_send.o zns_mgmt_recv.o channel_model.o conv_ftl.o simple_ftl.o pqueue/pqueue.o 

ccflags-$(CONFIG_NVMEVIRT_KV) += -DBASE_SSD=KV_PROTOTYPE
nvmev-$(CONFIG_NVMEVIRT_KV) += kv_ftl.o append_only.o bitmap.o
```

DingcuiYu, 2024-07-16:
Hi, I had the same problem and successfully complied NVMeVirt after following the above steps. Then, I set both `NS_CAPACITY_0`  and `NS_CAPACITY_1` as `GB(2)` . However, I only see one namespace after running `lsblk` command.

Are any other modifications required?

The following is the `dmesg` output:
```
[46383.964490] NVMeVirt: Version 1.10 for >> ZNS SSD Prototype <<
[46383.964521] NVMeVirt: Storage: 0x1640100000-0x1740100000 (4096 MiB)
[46383.978519] NVMeVirt: Total Capacity(GiB,MiB)=4,4224 chs=8 luns=32 lines=128 blk-size(MiB,KiB)=1,1056 line-size(MiB,KiB)=33,33792
[46383.993530] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.006249] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.018933] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.031609] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.044030] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.056173] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.068110] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.079919] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.079951] NVMeVirt: [chmodel_init] bandwidth 3050 max_credits 99 tx_time 40
[46384.079953] NVMeVirt: zone_size=33554432(Byte),32(MB), # zones=128 # die/zone=32 
[46384.079961] NVMeVirt: ns 0/2: size 4096 MiB
[46384.079965] NVMeVirt: Total Capacity(GiB,MiB)=0,0 chs=8 luns=32 lines=0 blk-size(MiB,KiB)=1,1056 line-size(MiB,KiB)=33,33792
[46384.079998] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080028] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080057] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080088] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080117] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080146] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080176] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080206] NVMeVirt: [chmodel_init] bandwidth 450 max_credits 14 tx_time 271
[46384.080234] NVMeVirt: [chmodel_init] bandwidth 3050 max_credits 99 tx_time 40
[46384.080236] NVMeVirt: zone_size=33554432(Byte),32(MB), # zones=0 # die/zone=32 
[46384.080238] NVMeVirt: ns 1/2: size 0 MiB
[46384.080301] PCI host bridge to bus 0001:10
[46384.080303] pci_bus 0001:10: root bus resource [io  0x0000-0xffff]
[46384.080307] pci_bus 0001:10: root bus resource [mem 0x00000000-0x3fffffffffff]
[46384.080310] pci_bus 0001:10: root bus resource [bus 00-ff]
[46384.080319] pci 0001:10:00.0: [0c51:0110] type 00 class 0x010802
[46384.080324] pci 0001:10:00.0: reg 0x10: [mem 0x1640000000-0x1640003fff 64bit]
[46384.080328] pci 0001:10:00.0: enabling Extended Tags
[46384.080734] NVMeVirt: Virtual PCI bus created (node 0)
[46384.081464] NVMeVirt: nvmev_io_worker_0 started on cpu 8 (node 0)
[46384.081568] NVMeVirt: nvmev_dispatcher started on cpu 7 (node 0)
[46384.081956] nvme nvme2: pci function 0001:10:00.0
[46384.090476] nvme nvme2: 1/0/0 default/read/poll queues
[46384.096594] NVMeVirt: Virtual NVMe device created
```

jaehshim, 2024-07-16:
From the `dmesg`, it seems that you have not specified the capacity of each namespaces.
You can specify the capacity of each namespaces at [ssd_config.h#L182](https://github.com/snu-csl/nvmevirt/blob/07acdd24e03d699bc88b272966d083bad21c82c6/ssd_config.h#L182)

DingcuiYu, 2024-07-16:
I modified `NS_CAPACITY_0` and `NS_CAPACITY_1`. However, it still doesn't work.

Is the way I modified them wrong?

Here are my modifications in `ssd_config.h`:
```
#define NS_SSD_TYPE_0 SSD_TYPE_ZNS
#define NS_CAPACITY_0 GB(2)
#define NS_SSD_TYPE_1 NS_SSD_TYPE_0
#define NS_CAPACITY_1 GB(2)
```

DingcuiYu, 2024-07-16:
I solved it! `NS_CAPACITY_0` and `NS_CAPACITY_1` should be set to GB (2ULL).

closed 2024-12-12 by beowulf