snu-csl/nvmevirt #45: Instructions to setup KVSSD with OpenMPDK/KVSSD

rwk-unil, 2024-07-03:
Hello,
Thank you for your work, I was interested in setting up the KVSSD emulated device but could not find information on how to do so on your GitHub. From the ACM ToS paper I found out that it was setup with the OpenMPDK (Samsung) KVSSD (https://github.com/OpenMPDK/KVSSD) driver as there is no NVMe KV driver in Linux yet.

Would it be possible to provide some basic instructions to replicate this setup ? Or link the instructions on the OpenMPDK project to test your KVSSD implementation ?

Thank you very much, this would be highly appreciated.

Best regards,
Rick

jaehshim, 2024-07-04:
We used Kernel device driver and kvbench for our evaluations.
Below are the links for detailed instructions on building the setup.
- [kernel device driver](https://github.com/OpenMPDK/KVSSD/tree/master/PDK/driver/PCIe/kernel_driver)
- [KVSSD library](https://github.com/OpenMPDK/KVSSD/tree/master/PDK/core)
- [kvbench](https://github.com/OpenMPDK/KVSSD/tree/master/application/kvbench)

Since the highest kernel version that OpenMPDK KVSSD repo provide is 4.15.18, we ported the kernel device driver to 5.10.37. (https://github.com/snu-csl/KVSSD_5.10.37)

rwk-unil, 2024-07-04:
Thank you very much for your reply and thank you for sharing the port of the driver to 5.10.37.

rwk-unil, 2024-07-08:
Just a quick question, does NVMeVirt KVSSD work with both the [kernel device driver](https://github.com/OpenMPDK/KVSSD/tree/master/PDK/driver/PCIe/kernel_driver) and the [user space driver](https://github.com/OpenMPDK/KVSSD/tree/master/PDK/driver/PCIe/user_driver) or only the kernel driver ?

Thank you.

jaehshim, 2024-07-10:
We have used the kernel device driver for the evaluations in the paper.
However, I think the user space driver should also work.

Please let us know if you face any problems while using the user space driver.

rwk-unil, 2024-07-12:
Thank you, I managed to run the kv_bench program from OpenMPDK with the kernel device driver and NVMeVirt as the KV SSD. It could be nice to have the exact commands you ran for the paper, in order for other researchers to replicate the benchmarks and compare the results from their machine.

Thanks

jaehshim, 2024-07-16:
From [bench_config.ini](https://github.com/OpenMPDK/KVSSD/blob/master/application/kvbench/bench_config.ini), I have modified the fixed_size in [key_length] and [value_length] to change the KV pair size. 
And depending on the workload you want to make, modify the read_write_insert_delete. 

If you face a error regarding the value_pool_unit, you may need to change the key_pool_unit and the value_pool_unit using the fixed_size in [key_length] and [value_length]

closed 2024-07-24 by jaehshim