snu-csl/nvmevirt #46: KV command delete not implemented ?

rwk-unil, 2024-07-12:
Hello,
When running NVMeVirt with the KV SSD configuration I found out that the "delete" command was not implemented.

I receive the error message `command not implemented` from https://github.com/snu-csl/nvmevirt/blob/main/kv_ftl.c#L990

I can see the code for `delete` does exist e.g., here https://github.com/snu-csl/nvmevirt/blob/main/kv_ftl.c#L564

Is there a reason why the delete command is not supported ? Does the delete code work ?

For example one could add the following in `kv_ftl.c` :

```diff
diff --git a/kv_ftl.c b/kv_ftl.c
index e4439e5..35ea418 100644
--- a/kv_ftl.c
+++ b/kv_ftl.c
@@ -978,6 +978,8 @@ bool kv_proc_nvme_io_cmd(struct nvmev_ns *ns, struct nvmev_request *req, struct
                ret->nsecs_target = __schedule_flush(req);
                break;
        case nvme_cmd_kv_store:
+       case nvme_cmd_kv_delete:
+               ret->nsecs_target = __schedule_flush(req);
        case nvme_cmd_kv_retrieve:
```

This will then call the delete code, is there a reason it is not activated ? Is is possible the delete code has a bug or would generate issues on the PC running the NVMeVirt module ?

Thank you very much.
Regards,
Rick

jaehshim, 2024-07-16:
It seems that the error log confused you.
Currently the error log is somewhat misleading.

In [kv_ftl.c#L990](https://github.com/snu-csl/nvmevirt/blob/main/kv_ftl.c#L990), it means that the performance model is not implemented for the specific opcode. 
This means that the nsecs_target == nsecs_start, and the command completion will happen when the actual delete operation finishes. (kv_ftl.c#L564).

rwk-unil, 2024-07-16:
Thank you, I understand, the error message was what was confusing me indeed.
I see now that this is for the performance logs (`proc_io_cmd()`) and not for the execution of the commands (`perform_io_cmd()`).
Thanks

Best regards,
Rick

closed 2024-07-16 by rwk-unil