snu-csl/nvmevirt #49: Is the aggregating write I/O in __zns_write wrong?

DingcuiYu, 2024-07-23:
In Conventional SSD, when determining whether the aggregated I/O threshold has been reached, use `ppa.g.pg % spp->pgs_per_oneshotpg == spp->pgs_per_oneshotpg - 1` ( [conv_ftl.c:12](https://github.com/snu-csl/nvmevirt/blob/07acdd24e03d699bc88b272966d083bad21c82c6/conv_ftl.c#L12)). 
However, in ZNS SSD, use `ppa.g.pg + pgoff == spp->pgs_per_pneshotpg` （[zns_read_wrire.c:186](https://github.com/snu-csl/nvmevirt/blob/07acdd24e03d699bc88b272966d083bad21c82c6/zns_read_write.c#L186C48-L186C50)).
It looks like the one in Conventional SSD is correct, but the one in ZNS SSD is wrong?

abcinje, 2024-07-24:
In the ZNS SSD model, the first condition of the if-statement at `zns_read_write.c#L186` is `(pg_off + pgs) == spp->pgs_per_oneshotpg`. (different from what you've mentioned)
The condition checks whether the sum of the page offset aggregated so far and the amount of data about to be written equals to the size of a oneshot chunk.

It seems to have no problems. Can you please let us know the wrong point in detail?

DingcuiYu, 2024-07-24:
Sorry, I realized I misunderstood this part of the code. Thank you very much for your reminder and explanation.

closed 2024-07-24 by DingcuiYu