snu-csl/nvmevirt #57: emulate 64T nvme SSD

mynvme, 2025-01-08:
Hello, 
Thanks for your contribution,  I was interested in setting up a **64T** nvme SSD, but I just have **64G** RAM memory on my ubuntu 20.04, linux 5.15.0-119 kernel server，how can I do this? Does simulating an SSD of a certain capacity require an equal amount of memory space？Thanks a lot!

jaehshim, 2025-01-09:
Hi, 

Yes. You need an equal amount of memory to emulate an SSD of a certain capacity.
However, if the actual data stored in the SSD doesn't matter, you can configure the FTL's mapping to cover that capacity while storing data in a modular fashion (e.g., LBA % size of memory space).

mynvme, 2025-01-09:
Hi @jaehshim ,
Thanks for your quick response and clarification, other than LBA % size method, is it a better solution to map the memory data to the backend disk files(each size 64G) based on the LBA / size? which can save all the data. The weakness is needing to map/unmap many times.

jaehshim, 2025-01-13:
Your method can also work as long as the 64T NVMe SSD that you are emulating is slower than the backend disk.

closed 2025-01-17 by mynvme