snu-csl/nvmevirt #58: when enable/disable controller frequently, admin complete queue may does not work

iamwangmali, 2025-02-21:
if we enable/disable the controller frequently(without shutdown the controller, and without update admin queue attribute), admin complete queue may does not work, as when enable controller, the field 'phase'  'cq_head' in admin queue, dbs[] in nvmev_vdev  does not reset

iamwangmali, 2025-03-05:
sorry to reply late, it maybe only occurred to me, as I am new learner of NVMe, and I port the code to QEMU, and to replace some code there.  when booting，after the Ctrl had been reseted,  AQA was not changed before enable the ctrl once more.
Thanks for your reply.

beowulf, 2025-03-14:
@iamwangmali Thanks for your response. I will check the part and see if there is any issue there.

closed 2025-03-14 by beowulf