snu-csl/nvmevirt #60: Why do I fail to reserve memory?

DingcuiYu, 2025-02-24:
I try to reserve memory by editing Linux command-line configuration in `/etc/default/grub`:
```
GRUB_CMDLINE_LINUX="memmap=79G\\\$82G isolcpus=7,8"
```
And I update and reboot the machine using:
```
sudo update-grub
sudo reboot
```
In the output of  `dmesg` , I can see:
```
[    0.000000] user: [mem 0x0000001480000000-0x000000283fffffff] reserved
```
However, these memory is not successfully reserved because as shown in   `/proc/iomem` , part of these memory is occupied by kernel code and kernel data:
```
1480000000-1aab3fffff : Reserved
1aab400000-1aae42ffff : System RAM
  1aab400000-1aac7fffff : Kernel code
  1aac800000-1aad54cfff : Kernel rodata
  1aad600000-1aada596ff : Kernel data
  1aadf46000-1aae3fffff : Kernel bss
1aae430000-283fffffff : Reserved
```
How can I protect these memory from kernel's occupation?

DingcuiYu, 2025-02-24:
I use the following command and the problem seems to be solved:
```
GRUB_CMDLINE_LINUX="mem=82G memmap=79G\\\$82G isolcpus=7,8"
```
It seems that the `mem` parameter sets a memory boundary that can be used. But I'm not sure if this is the solution to the problem, since the kernel data and code appear in a different place each time.

DingcuiYu, 2025-02-25:
Here is the boot log:
```
[    0.000000] Linux version 6.8.0+ (lab@server3) (gcc (Ubuntu 9.5.0-1ubuntu1~22.04) 9.5.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP PREEMPT_DYNAMIC Thu Dec 12 23:05:44 CST 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.8.0+ root=UUID=ace14467-43e4-4018-8e27-3dea88bc4773 ro memmap=79G$82G isolcpus=7,8 quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000005cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000005d000-0x000000000005efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000005f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000068792fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000068793000-0x0000000069f7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000069f7e000-0x000000006a9adfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000006a9ae000-0x000000006abbffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000006abc0000-0x000000006f7fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000006f800000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000287fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] APIC: Static calls initialized
[    0.000000] e820: update [mem 0x66231018-0x66264a57] usable ==> usable
[    0.000000] e820: update [mem 0x66231018-0x66264a57] usable ==> usable
[    0.000000] e820: update [mem 0x661fd018-0x66230a57] usable ==> usable
[    0.000000] e820: update [mem 0x661fd018-0x66230a57] usable ==> usable
[    0.000000] e820: update [mem 0x661dc018-0x661fc457] usable ==> usable
[    0.000000] e820: update [mem 0x661dc018-0x661fc457] usable ==> usable
[    0.000000] e820: update [mem 0x66462018-0x6646c657] usable ==> usable
[    0.000000] e820: update [mem 0x66462018-0x6646c657] usable ==> usable
[    0.000000] e820: update [mem 0x65548018-0x65552657] usable ==> usable
[    0.000000] e820: update [mem 0x65548018-0x65552657] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000005cfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000005d000-0x000000000005efff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000005f000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000065548017] usable
[    0.000000] reserve setup_data: [mem 0x0000000065548018-0x0000000065552657] usable
[    0.000000] reserve setup_data: [mem 0x0000000065552658-0x00000000661dc017] usable
[    0.000000] reserve setup_data: [mem 0x00000000661dc018-0x00000000661fc457] usable
[    0.000000] reserve setup_data: [mem 0x00000000661fc458-0x00000000661fd017] usable
[    0.000000] reserve setup_data: [mem 0x00000000661fd018-0x0000000066230a57] usable
[    0.000000] reserve setup_data: [mem 0x0000000066230a58-0x0000000066231017] usable
[    0.000000] reserve setup_data: [mem 0x0000000066231018-0x0000000066264a57] usable
[    0.000000] reserve setup_data: [mem 0x0000000066264a58-0x0000000066462017] usable
[    0.000000] reserve setup_data: [mem 0x0000000066462018-0x000000006646c657] usable
[    0.000000] reserve setup_data: [mem 0x000000006646c658-0x0000000068792fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000068793000-0x0000000069f7dfff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000069f7e000-0x000000006a9adfff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000006a9ae000-0x000000006abbffff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000006abc0000-0x000000006f7fffff] usable
[    0.000000] reserve setup_data: [mem 0x000000006f800000-0x000000008fffffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000147fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000001480000000-0x000000283fffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000002840000000-0x000000287fffffff] usable
[    0.000000] user-defined physical RAM map:
[    0.000000] user: [mem 0x0000000000000000-0x000000000005cfff] usable
[    0.000000] user: [mem 0x000000000005d000-0x000000000005efff] reserved
[    0.000000] user: [mem 0x000000000005f000-0x000000000009ffff] usable
[    0.000000] user: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] user: [mem 0x0000000000100000-0x0000000065548017] usable
[    0.000000] user: [mem 0x0000000065548018-0x0000000065552657] usable
[    0.000000] user: [mem 0x0000000065552658-0x00000000661dc017] usable
[    0.000000] user: [mem 0x00000000661dc018-0x00000000661fc457] usable
[    0.000000] user: [mem 0x00000000661fc458-0x00000000661fd017] usable
[... 1952 lines cut ...]
[   13.003354] intel_rapl_common: Found RAPL domain package
[   13.003367] intel_rapl_common: Found RAPL domain dram
[   13.014677] intel_rapl_common: Found RAPL domain package
[... cut ...]
[   28.211275] rfkill: input handler disabled
[   54.173880] audit: type=1400 audit(1740410517.285:55): apparmor="DENIED" operation="capable" class="cap" profile="/snap/snapd/23545/usr/lib/snapd/snap-confine" pid=1982 comm="snap-confine" capability=12  capname="net_admin"
[   54.173900] audit: type=1400 audit(1740410517.285:56): apparmor="DENIED" operation="capable" class="cap" profile="/snap/snapd/23545/usr/lib/snapd/snap-confine" pid=1982 comm="snap-confine" capability=38  capname="perfmon"
```

beowulf, 2025-02-27:
@DingcuiYu 
I see several WARN(), BUG(), and errors in the bootlog....
The reported case does not happen on any of our systems. So it would be better to rule out first that they are not the cause of the issue. Would you like to test with the vanilla kernel booting with the memmap option but without isolcpus, and check kernel addresses?

DingcuiYu, 2025-02-27:
@beowulf 
After comparing bootlogs under four permutations of vanilla Linux/modified Linux,  with and without adding the `isolcpus`, I think it's the modified Linux that should be causing the problem.
Comparing the bootlog I found the following warning prompted:
```
[ 0.000197] .text .data .bss are not marked as E820_TYPE_RAM!
[ 0.000198] e820: remove [mem 0x1c24a00000-0x1c27a2ffff]
```

`0x1c24a00000-0x1c27a2ffff` is the memory occupied by kernel code and kernel data.

I just added a new member to `struct bio`, and modified files like `block/bio.c`,`block/bounce.c`,`driverse/nvme/host/core.c`,`fs/f2fs/data.c`,`include/linux/blk_types .h`, don't quite understand how this actually affects memory allocation at kernel startup. I know this is probably beyond the scope of your answer, but if you could give me some advice it would be greatly appreciated. :-)

arter97, 2025-02-27:
The bootloader or the kernel's EFI stub (which is used by the bootloader) ultimately decides where the Linux kernel gets uploaded.

If KASLR is active (which it is by default), your kernel can be uploaded anywhere in the memory. A (relatively) recent commit from v6.10 fixes this behavior: https://lore.kernel.org/lkml/202405161154.01864575AD@keescook/T/

If you don't have that commit, the bootloader can keep picking up random address to upload the kernel, which explains why sometimes your memmap works and sometimes not.

That commit was backported to the stable tree (which in turn would be picked by Canonical's Ubuntu tree for non-LTS backports).

I see that you're running v6.8.0. stable/linux-6.8.y did not receive the backport and went EOL first. If you forked from Canonical's tree, you'd need at least `6.8.0-45`: https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/noble/commit/?id=91b0b2b4320cc98b1e9c3218970a84dc49ffcf9d

I encourage you to pick a reliable LTS kernel version, either v6.6.80 or v6.12.16. If you absolutely need v6.8, pick Canonical's latest v6.8 kernel from noble: https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/noble

closed 2025-02-27 by DingcuiYu

beowulf, 2025-02-28:
@arter97 Been a while! Thanks for the information, you saved me once again! 🤣🤣
@DingcuiYu Good to hear the issue is resolved!