snu-csl/nvmevirt #63: It seems that free_seq sometimes catch up with free_seq_end.

cactoid, 2025-05-01:
I am using an 8 core machine, two cores are allocated for NVMeVirt (one for dispatcher and the other for io_worker), with CONFIG_NVMEVIRT_NVM setting, running small size (=512-Byte) random reads. I control NVMe directly from userspace and use 2 NVMe SQs, each queue has a maximum QD=512.

In __allocate_work_queue_entry(), I sometimes see a warning corresponding to the conditional statement "if (w->next >= NR_MAX_PARALLEL_IO)". In that case, w->next was -1. Also, as far as I checked, it seems that free_seq sometimes catches up with (= is equal to) free_seq_end, which may be the cause of the problem, though I don't understand the whole structure of NVMeVirt.

NR_MAX_PARALLEL_IO is the default 16K, which is much larger than QD=1024 for my two SQs in total, but I guess that if there is an entry whose reclaim is delayed due to, for example, having only one io_worker thread for two SQs (, or any other reason), such catch-up may occur.

My two questions are as follows.
 (1) It may be a corner case, but can free_seq catch up with free_seq_end as above?
 (2) Is there any way to prevent the above warning from occurring?  (other than increasing NR_MAX_PARALLEL_IO to a larger value or increasing the io_worker thread because I don't think they are fundamental solutions.)

beowulf, 2025-05-01:
Could you check if your userspace program handles cq properly?

cactoid, 2025-05-07:
Thanks for your quick reply. I've checked my CQ handling. It looks OK, although I'm not 100% sure.

As far as I examine my execution log, I suspect some corner case like the following is happening.

First, during execution, the __reclaim_completed_reqs() function observes w->is_completed = false in some nvmev_io_work w pointed to by worker->io_seq i.e. first_entry, and the corresponding last_entry becomes -1.

Next, we assume the case where nvmev_io_worker() starts from the same worker->io_seq as above at about the same time and handles the above w, but (w->nsecs_target <= curr_nsecs) does not hold and w->is_completed still remains false.

nvmev_io_worker() processes next item w' as far as w->next can be tracked, and returns it to CQ (w'->is_completed may become true).
If requests are added to SQ fast enough, nvmev_io_worker can keep processing them without exiting the while (curr != -1) loop.
Reclaim will keep looking at the same io_seq, and first_entry's is_completed remains false (in the case that w is not processed again in nvmev_io_worker) and reclaim will not proceed.
In this way, the work_queue can be used up and free_seq can become equal to free_seq_end.